earlephilhower/arduino-esp8266littlefs-plugin #1: LittleFS Error: mklittlefs not found!

DiegoRodasS, 2019-07-06:
Hello Earl,
I'm getting (LittleFS Error: mklittlefs not found!) when trying to upload files to the esp, there is already created the data directory and a .html file inside the directory.

earlephilhower, 2019-07-06:
You need the mklittlefs tool  It should be installed by the IDE automatically, but if you are using GIT, then you need to run `python get.py` in the `tools` directory to download it.

DiegoRodasS, 2019-07-06:
Excuse me,  I'm working on windows nt, I don't know what's GIT and don't know anything about python.

DiegoRodasS, 2019-07-06:
I tried what you told me, but I got this :

$ PATH=$PATH:/c/Python27/
$ python get.py
C:\Python27\python.exe: can't open file 'get.py': [Errno 2] No such file or directory
So, do I have to install the core with git to solve the issue?

earlephilhower, 2019-07-06:
The get.py script is in your IDE clone location in the `tools` directory.  It's not a python built-in, of course, so you need to be in that dir to make it work...

DiegoRodasS, 2019-07-06:
As I didn't install the core using git... I'll reinstall the Core from the IDE, if it doesn't work, then the IDE and the library, so on íf it doesn't, I'll clone the repositiry with GIT.

earlephilhower, 2019-07-06:
Ah, that explains it!  2.5.2 did not have LittleFS support, it was only bugfixes for 2.5.0.  You need to use GIT installs if you want to try littleFS, or wait for 2.6.0...

earlephilhower, 2020-02-10:
Closing as LittleFS has been in the core toolset for a long time now.

closed 2020-02-10 by earlephilhower