earlephilhower/arduino-esp8266littlefs-plugin #3: Plugin not found

mike-s123, 2019-12-03:
Windows 7 64bit
Arduino IDE 1.8.10
ESP8266 Platform 2.7.0 (via git)
LittleFS plugin 2.6.0

IDE does not find the littlefs upload tool, it never appears in the IDE 

It does find the spiffs tool, which is installed in the same directory (~/Documents/Arduino/tools/ESP8266FS/tool). I can install/delete the spiffs tool and it comes and goes in the IDE, so I'm sure I'm placing them in the proper place. Both have the same permissions.

I did run get.py from inside ~/Documents/Arduino/hardware/esp8266com/esp8266/tools, but I don't think that has anything to do with the plugin not being found.

`$ python get.py
Platform: x86_64-mingw32
Tool python3-3.7.2.post1-embed-win32v2a.zip already downloaded
Extracting dist/python3-3.7.2.post1-embed-win32v2a.zip
Renaming python3-3.7.2.post1-embed-win32/ to python3
Tool x86_64-w64-mingw32.xtensa-lx106-elf-b40a506.1563313032.zip already downloaded
Extracting dist/x86_64-w64-mingw32.xtensa-lx106-elf-b40a506.1563313032.zip
Tool x86_64-w64-mingw32.mkspiffs-7fefeac.1563313032.zip already downloaded
Extracting dist/x86_64-w64-mingw32.mkspiffs-7fefeac.1563313032.zip
Downloading x86_64-w64-mingw32-mklittlefs-69bd9e6.zip
Done
Extracting dist/x86_64-w64-mingw32-mklittlefs-69bd9e6.zip
`

earlephilhower, 2019-12-03:
You need to install the plug in manually, like the spiffs uploader.  The readme explanation the proper spot to install.

closed 2019-12-03 by earlephilhower

mike-s123, 2019-12-03:
Yes, that's exactly what I'm doing.

     ~/Documents/Arduino/tools/ESP8266FS/tool
    $ ls -la
    total 16
    drwxr-xr-x 1 mike-s 1049089    0 Dec  3 12:12 ./
    drwxr-xr-x 1 mike-s 1049089    0 Jan 20  2019 ../
    -rw-r--r-- 1 mike-s 1049089 7131 Dec  3 10:33 esp8266fs.jar
    -rw-r--r-- 1 mike-s 1049089 7207 Dec  3 10:53 esp8266littlefs.jar

As I said, the spiffs uploader is found and shown in the IDE menu, the littlefs one is not. Yes, I've quit/restarted the IDE (many times).