earlephilhower/arduino-esp8266littlefs-plugin #6: OTA Authentication fail.

cmbhjv, 2020-02-09:
When trying to do OTA in a device with OTA password set, authentication fails.

It doesn't ask for the password either.

My guess is that it needs to pop the Password request form (I guess it's this one [here](https://github.com/arduino/Arduino/blob/master/app/src/processing/app/forms/PasswordAuthorizationDialog.java) ) and pass it as the "-a" argument to espota.py ([here](https://github.com/earlephilhower/arduino-esp8266littlefs-plugin/blob/aef8eceef43205fab21bb870d7883c14bf88fc42/src/ESP8266LittleFS.java#L323))

earlephilhower, 2020-02-10:
This is the mklittlefs.exe github, not related to OTA.  I suggest you look at the esp8266 core for assistance.

cmbhjv, 2020-02-12:
Sorry, but I believe you are mistaken. mklittlefs.exe is [another repository](https://github.com/earlephilhower/mklittlefs).<br>

This is the **"arduino-esp8266littlefs-plugin"** github. 
[image]

It's the arduino-esp8266littlefs-plugin who needs to ask for the password. The ESP8266LittleFS.java file in **_this_** repository [calls ](https://github.com/earlephilhower/arduino-esp8266littlefs-plugin/blob/aef8eceef43205fab21bb870d7883c14bf88fc42/src/ESP8266LittleFS.java#L323)the espota.py without the -a argument.

earlephilhower, 2020-02-12:
Ah, you are correct.  I do own both repos.  However, the littlefs-java uploader is mostly a cut-n-paste version of the standard arduino 8266 one by me-no-dev:  

https://github.com/esp8266/arduino-esp8266fs-plugin/blob/56a6c515545ff5301be5e679d014ab7062600d81/src/ESP8266FS.java#L322

I don't believe it gets a notice that the OTA has a password (most don't since OTA is insecure and will pass PWDs plaintext over the air).  Not sure there's a way to know if a PWD has been set in the IDE.  Will need a little study.

If you've got a suggestion, PRs are always appreciated. :)

877dev, 2020-09-23:
I've been having the same issue here, ArduinonIDE does not prompt for a password when uploading LittleFS OTA. 

A workaround is to flash a sketch with no password, upload LittleFS, then flash sketch with password.

LeisureLadi, 2021-07-18:
cmbhjv and 877dev,
I have forked the repo and added a dialog box for the password. The downside is, that the box will open every time, no matter if a password is required or not. But you can leave the box empty to upload without a password.
Have fun!

877dev, 2021-07-18:
> cmbhjv and 877dev,
> I have forked the repo and added a dialog box for the password. The downside is, that the box will open every time, no matter if a password is required or not. But you can leave the box empty to upload without a password.
> Have fun!

That sounds good thank you. I’ll check that out and report back! 👍

877dev, 2021-07-26:
@LeisureLadi @cmbhjv finally I got to test this (been away for a week).
It works!

 Password box appears, enter OTA password, small delay, then `Authenticating...OK` and `Uploading....`. Takes a while to upload, and there is no `Finished` message. But is works.

Excellent work, thank you.

@earlephilhower FYI https://github.com/earlephilhower/arduino-esp8266littlefs-plugin/pull/18

LeisureLadi, 2021-07-29:
@877dev 
The missing 'Finished' message is related with the Python script. It is missing there. Having said this, it is also missing, when you OTA a sketch. I'm not sure if it can be added there. Feel free to investigate and let us know, what you find.

MxMantoHI, 2022-02-20:
Hi bros. Scuse me, where i can download .jar with the Password box update?
commented are on jul 2021, but release are on 18 Nov 2019. regards!!