earlephilhower/arduino-esp8266littlefs-plugin #9: This plugin vs. mklittlefs compatibiliy. LittleFS root files slash

lorol, 2020-06-03:
Hi, 
By using your other tool mklittlefs, I tried to unpack an image made by this plugin with Arduino IDE (Win10 64). The /data folder of the sketch has an inside folder.
mklittlefs.exe -l -s 2072576 MyProject.mklittlefs.bin
lfs error:966: Corrupted dir pair at 130 131
If I use mklittlefs to pack and unpack, it is OK. 
Are both compatible and what parameters I need to provide (size, block) to be on "same page" between them?

My other question is about leading /  of root files of LittleFS.
I replaced relatively easy SPIFFS with LittleFS in this project:
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp
It works ... 
However with LittleFS it somehow "strips" the leading slashes of root files and I need to modify other things, example "/.exclude.files" -> ".exclude.files"
Is this anything to do with notes here:
https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#littlefs-file-system-limitations

Or it is a difference that cannot / should not be retrofitted to look exactly like with SPIFFS?

earlephilhower, 2020-06-03:
The SPIFFS one is easy.  There is no such thing as a directory on SPIFFS.  That first "/" (and all others) is actually part of the file name (i.e. a file can be called "/this//is/a///file").  "/file.txt" is a different file than "file.txt" in this case.

LittleFS has directories so "file" and "/file" are the same file.  For most things, I imagine it will just work with the brain-damaged SPIFFS user code.

As to the mklittlefs, you need to specify the exact same parameters on the `-l` command line as the Arduino GUI sends to it (block, page, total size).  You can check the source here to find the param variables it uses, and then check your boards.txt for your specific chip to get the right values for them.  OTW, it uses defaults which are guaranteed not to match your chip exactly...

lorol, 2020-06-03:
Ok I got that :) 
So  LiteFS drivers play with interpreting of these slashes.
All good. Thanks.

BTW, to be able to build on Arduino IDE, I needed to checkout (insert manually)
https://github.com/ARMmbed/littlefs 
to the otherwise empty:
C:\Users\<username>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.<x.x>\libraries\LittleFS\lib\

Is this supposed to be that way?

And also BTW, i did my very quick-n-dirty first replacement SPIFFS with LittleFS test in my sketch this (ridiculous) way, which worked:
...
#include <FS.h>
// #include <SPIFFS.h>
#define SPIFFS LittleFS
#include <LittleFS.h>
...

Do you have anything similar for LittleFS but for ESP32?

Thank you for your great work, appreciated!

earlephilhower, 2020-06-03:
I think your install got borked.  The 2.7.1 package ZIP that the board manager uses ( https://github.com/esp8266/Arduino/releases/download/2.7.1/esp8266-2.7.1.zip ) does include that subdir with source, I just checked it.

If you're using `git` then you need the usual `git submodule init` and `git submodule update`.

I unfortunately don't have an ESP32 version.  They've got the `vfs` filesystem stuff.  It should be easy enough for anyone to port.  I made 0 changes to the ARM maintained FS code, just had to put wrappers around it.

For most things, your `#define` change will work, but you do need to be aware there are now directories if you iterate over things.

Good luck!

closed 2020-06-03 by earlephilhower

lorol, 2020-06-11:
Hi @earlephilhower 
Thank you for the advice - I fixed my install and now all compiles fine.
And the LittleFS works great with very minor changes of existing code that uses SPIFFS regards of directories and slashes.

Now, hoping one day to see a ESP32 implementation with that similar great compatibility to ESP32 SPIFFS
Seems to be not so easy. No one has it fully done yet for Arduino. Will wait.
For me, it looks too complex or maybe impossible to bring it through ESP-IDF (VFS?) to Arduino.

Meanwhile I made an improvised "arduino-esp32littlefs-plugin" see attached, but is pretty useless :) before we have a lib.

[arduino-esp32littlefs-plugin.zip](https://github.com/earlephilhower/arduino-esp8266littlefs-plugin/files/4766351/arduino-esp32littlefs-plugin.zip)

lorol, 2020-06-17:
https://github.com/espressif/arduino-esp32/issues/3765#issuecomment-645486244

I made it for ESP32