earlephilhower/arduino-esp8266littlefs-plugin #10: How to increase filename limit from 32?

BRMateus2, 2020-08-11:
Hello, I did study about LittleFS and its limitations of 31 bytes + 1 (/0) characters, the problem is that some website files are hardcoded (or unlikely to change), for example:
```
/fontawesome/webfonts/fa-solid-900.woff2
*** buffer overflow detected ***: terminated
```

I already changed `#define LFS_NAME_MAX 32`, from the core library, to the needed value, but I still cannot simply upload those files (because it is not dinamically detected) and debug for larger file paths.

Is it possible to make LittleFS more flexible for beginner devs?

earlephilhower, 2020-08-16:
@BRMateus2, that 32 is the length of a path element (i.e. dirname or filename).  You should be able to string multiple elements up to 256b (IIRC).  If not, there's a bug in the mklittlefs code because the Arduino core definitely supports multiples of 32b...

Let me take a look...

BRMateus2, 2020-08-16:
I had exactly that problem you described :), using Linux Fedora with Arduino 1.8.13 - for some reason then, it counted the whole path as maximally 32 bytes (the above path has 40 characters).

earlephilhower, 2020-08-16:
I was able to repro the failure w/the info you provided.  I've committed a simple fix (had LFS_FILE_MAX instead of PATH_MAX in a temp variable) to the main mklittlefs repo.

You'll need to pull the latest source from https://github.com/earlephilhower/mklittlefs/ , build, and then install the executable in the tools/mklittlefs/. path.  Should be good to go, now.

closed 2020-08-16 by earlephilhower