earlephilhower/arduino-esp8266littlefs-plugin #15: LittleFS Error: mklittlefs not found!

jancoow, 2021-02-03:
Hi,

I did exactly what is in the readme, however, after I click on "ESP8266 LittleFS data upload" I get this error:

> LittleFS Error: mklittlefs not found!

Which is weird, because I did install mklittlefs:

> mklittlefs --version
>    mklittlefs  version: 0.2.3-31-g295fe9b

busel7, 2021-05-19:
Same here, after update ESP8266 core v3.0.0, cant Data upload with LittleFS

Pius171, 2021-05-19:
> Same here, after update ESP8266 core v3.0.0, cant Data upload with LittleFS

3.0 is still in beta stage, I don't think it allows plugins for now, use a stable version

busel7, 2021-05-19:
> a stage, I

Ok, thank you for info

ESPboy-edu, 2021-05-29:
Same, after update ESP8266 core v3.0.0, cant Data upload with LittleFS

alvikskor, 2021-06-02:
The branch ESP8266 Core 3.0.0 has a fix
"Re-add mklittlefs to package.json (# 8060)", but apparently it didn't help. Yesterday (06/01/21) I tried to install the Core 3.0.0  using the "Board Manager". When trying to load the LittleFS file system into the board, the same error "littlefs error: mklittlefs not found!" occurred. The SPIFFS file system boots normally. Are there any ways to fix the problem.

earlephilhower, 2021-06-02:
The updated core is not yet released with patch.  Until then, please use the git version.

hoekbrwr, 2021-06-08:
Version 1.8.15 has this error. I am not able to use LITTLEFS anymore. The new program is now a .java program. How to restore the old procedures? Should I uninstall v3.0.0? All files are in place the .jar file is still there, but it needs this mklittlefs which I cannot find. As this is now the standard filesystem I expect to have a standard procedure to upload files to the filesystem in that format. Why this hocus-pocus via external contributors????

djbiton, 2023-04-19:
'LittleFS' was not declared in this scope

LITTLEFS Error: mklittlefs not found!