earlephilhower/arduino-esp8266littlefs-plugin #16: LIttleFS plugin not working on big sur

Pius171, 2021-04-24:
i followed the instructions in your github page but the littlefs option doesn't show up in arduino tools option

earlephilhower, 2021-04-24:
The plugin is plain java and works on Big Sur, but needs to be in the right directory for the IDE to find and run it.

Please double check that you have expanded the zip into a `<ArduinoSketchDir>/tools/PicoLittleFS/tool/picolittlefs.jar` file.
````
$ ls -l ~/Arduino/tools/PicoLittleFS/tool/picolittlefs.jar 
-rw-r--r-- 1 earle earle 6715 Mar 30 23:00 /home/earle/Arduino/tools/PicoLittleFS/tool/picolittlefs.jar
````

closed 2021-04-24 by earlephilhower

Pius171, 2021-04-24:
There is no picolittlefs directory in the expanded zip folder

earlephilhower, 2021-04-24:
For the ESP8266 version, it will be a ESP8266LittleFS dir no extraction.  For the RPI Pico, it will be the Pico one.

Pius171, 2021-04-24:
[image]

[image]

it still doesnt work. i am sorry i am disturbing you but i can't find any solution on the internet

earlephilhower, 2021-04-24:
Looks like the right spot, but you're running the 2.0 beta.  Maybe it doesn't work with plugins?  Try 1.8.xxx if you can.