earlephilhower/arduino-esp8266littlefs-plugin #20: ESP8266LittleFS does not show in Arduino

michalrusinowski, 2022-05-13:
Dears, 

I followed the steps you mentioned and I can see no LittleFS in tools.

System: MacOS
Arduino Sketchbook: 
Version: 2.0.0-rc6
Date: 2022-04-20T10:22:58.789Z (3 weeks ago)

I have created folder "tools" over ~/Documents/Arduino
and placed  ESP8266LittleFS there  so I have the path:
~/Documents/Arduino/tools/ESP8266LittleFS/tools/esp8266littlefs.jar.
I did restart Arduino... and nothing.
I tried to play a little putting the LittleFS folder in libraries or over aproject folder and nothing happens...

Any idea what could be wrong?
Thanks in advance

kai-rdh, 2022-05-31:
Arduino IDE 2.0 does not support plugins. You have to use 1.8 instead. 

https://github.com/arduino/arduino-ide/issues/58

earlephilhower, 2022-05-31:
Thanks @kai-rdh .  There's nothing we can do as end users, only the Arduino folks can enable this.

closed 2022-05-31 by earlephilhower