earlephilhower/arduino-esp8266littlefs-plugin #21: arduino-esp8266littlefs-plugin on the new Arduino IDE 2.0

ianburton20, 2022-09-19:
Do you plan to port this (very useful) tool to the new Arduino IDE 2.0 ? Thanks.

earlephilhower, 2022-09-19:
Tracked https://github.com/arduino/arduino-ide/issues/58 

Right now, it seems like it is not possible in the 2.0 IDE.

closed 2022-09-19 by earlephilhower

ianburton20, 2022-09-19:
Shame!

 

Thanks for the speedy response.

 

Tracked arduino/arduino-ide#58 <https://github.com/arduino/arduino-ide/issues/58>

HeinzRuetschi, 2023-02-02:
Hello earlephilhower, thank you very much for your super-fast reply, really a fantastic service. 
However, I was hoping somewhat for a more positive answer. Which leaves me with the question, is there another way to implement or apply your LittleFS data transfer with the Arduino IDE 2.0.3? Thank you again your help and great work.

ianburton20, 2023-02-02:
ElegantOTA might offer an option of LittleFS file upload. Not actually tried it - but noticed that is looks to have this capability - for ESP32 and ESP8266. Option outside of the Arduino IDE of course - but an option nevertheless.

HeinzRuetschi, 2023-02-02:
Thank you very much for your great support and fantastic service!

earlephilhower, 2023-02-02:
A manual workaround might be to use the 1.x IDE, select the proper chip and filesystem size, and do one upload.  The run log should show the command line the uploaded used for both `mklittlefs` and `uf2conv.py`.

You could then manually run those from a command line to do an upload as needed

HeinzRuetschi, 2023-02-03:
earlephilhower, super service! Thanks again for your great suggestions and incredible fast support.

HeinzRuetschi, 2023-02-03:
Gentlemen, with your great support I was now able to create a bin file of a data directory and upload the created bin file to the ESP8266 with ElegantOTA webpage as a server implemented and the use of the Arduino IDE 2.0.3.

As a test I use the project of random nerd tutorials, https://randomnerdtutorials.com/esp8266-nodemcu-ota-over-the-air-arduino/#comment-815024

I found the mklittlefs.exe on my system in the Arduino directory at 
C:\Users\HR\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec\
and used the following parameters in a terminal window:

mklittlefs -c C:\Users\HR\Documents\Arduino\Projects\ESP8266-OTA-Web_Server_LED_V2\data  C:\Users\HR\Documents\Arduino\Projects\ESP8266-OTA-Web_Server_LED_V2\fs_data.bin 
which worked great.

To improve the method I copied mklittlefs.exe into a reachable path and 
tried this again in the project directory which included the data directory,
which I wanted to upload to the ESP8266 server with littlefs:
mklittlefs -c data  fs_data_01.bin

After that I used the IP-Adresse of ESP8266/update:
http://192.168.189.107/update to upload the bin file by selecting Filesystem of the ElegantOTA page.

Gentlemen thanks again for your great support. Without your help I would not have been able to resolve this issue so quickly!

In any case, if you have any other recommendation you are always welcome.

MicSG-dev, 2023-09-22:
> Arduino/arduino-ide rastreado #58
> 
> No momento, parece que isso não é possível no IDE 2.0.

@earlephilhower 
Hello! 
Here's a response that one of the Arduino IDE 2.0 developers gave:
>[...] so Arduino IDE 2.x had support for VS Code extensions all along, and in fact the stock Arduino IDE installation already >makes heavy use of extensions. No additional work is needed on Arduino IDE codebase for VS Code extension support.
>
>Even though the VS Code extension system makes it possible to do quite a lot in adding additional capabilities to Arduino >IDE, in order to accomplish the equivalent to what is done in some of the Arduino IDE 1.x "plugin" tools, the extension also >needs access to the Arduino-specific data known to Arduino IDE/Arduino CLI. The lack of access to such data has been the >technical blocker for creating VS Code extension replacements for the Arduino IDE 1.x "plugin" tools. That missing capability >was recently added: https://github.com/arduino/arduino-ide/pull/2110
>
>So from a technical standpoint we can close this issue as resolved. The outstanding task for Arduino is documenting:
>
>-The fact that adding additional capabilities to Arduino IDE by 3rd parties is possible through VS Code extensions.
>-The availability of Arduino state data to extensions via an API

I hope that soon we can have access to its extension in Arduino IDE 2.0.

https://github.com/arduino/arduino-ide/issues/58#issuecomment-1728532605