earlephilhower/arduino-esp8266littlefs-plugin #22: littlefs on linux

gordnhoo, 2022-09-23:
I have been trying to install littlefs on linux but it seems to not be working i have tried following the same tutorials as for mac os and windows  but none work is it linxu compatible?

earlephilhower, 2022-09-23:
All my development work is on Ubuntu, so yes it does work on Linux. 😄   It's built with Java, which is included with the Arduino IDE 1.x (note that 2.x is not supported...they have no plug in support as of now!)

Double check the installation directory.  You'll want to expand into `~/Arduino/tools/ESP8266LittleFS` and should have a file `~/Arduino/tools/ESP8266LittleFS/tool/esp8266littlefs.jar` expanded there.

gordnhoo, 2022-09-23:
ho yes my problem was because of arduino IDE V2 i just installed the oldest version and it works fine.
When do you think plugin support will arrive?
Thank you for the dam quick reply <3

closed 2022-09-23 by gordnhoo

earlephilhower, 2022-09-23:
Subscribe to https://github.com/arduino/arduino-ide/issues/58 , but I have no idea if/when support will be added, sorry!