earlephilhower/arduino-esp8266littlefs-plugin #25: Plugin tool not compatible with Arduino IDE 2.x

ramanvda, 2023-04-14:
Hello,

It's apparently a limitation from Arduino IDE 2.x

Source : https://forum.arduino.cc/t/tool-menu-esp8266-littlefs-filesystem-uploader-not-displaying/1037055/3

Unfortunately the "ESP8266 LittleFS Filesystem Uploader" tool is not compatible with Arduino IDE 2.x:

> [github.com/arduino/arduino-ide](https://github.com/arduino/arduino-ide/issues/58)
> [Missing support for external tools / plugins 19](https://github.com/arduino/arduino-ide/issues/58)
> opened 11:42PM - 12 Feb 21 UTC
> szerwi[ szerwi](https://github.com/szerwi)
> topic: code type: imperfection criticality: high
> **Describe the solution you'd like**
> I would like to propose support for extern[…](https://forum.arduino.cc/t/tool-menu-esp8266-littlefs-filesystem-uploader-not-displaying/1037055/3)
> If you have a [GitHub 2](https://github.com/) account, you can subscribe to that issue to get notifications of any new developments related to this subject.

For now, if you require this tool then you should use Arduino IDE 1.8.19. The download links for Arduino IDE 1.8.19 are available here:

[https://www.arduino.cc/en/software#legacy-ide-18x 92](https://www.arduino.cc/en/software#legacy-ide-18x)

You can have both Arduino IDE 1.x and 2.x installed at the same time, so if you like Arduino IDE 2.x, you can use that version normally, then switch over to 1.8.19 only when you have need to use "ESP8266 LittleFS Filesystem Uploader" or a similar Java-based tool.

Regards

closed 2023-04-14 by earlephilhower

Delitants, 2023-05-08:
What is the problem to fix this?

earlephilhower, 2023-05-08:
See the linked IDE issue.  It's out of our hands here.